UBC-Rocket/Whistler-Blackcomb-v2 #50: RTT on WB-FC-Firmware

davinbirdi, 2022-03-18:
Next time I'm on that board I will add this so that we can use RTT to debug at faster speeds than printfs.

I've set up the first commit on the DAQ firmware so it works nicely to transfer over.

Esouder, 2022-10-04:
As of [0ac6c7c](https://github.com/UBC-Rocket/Whistler-Blackcomb-v2/commit/0ac6c7ce2630989719948458276969c4a5356cf7), RTT builds on MCUXpresso for WB-FC. Still need to test if it works and make sure that it hasn't broken x86 compilation.

Esouder, 2022-10-04:
I think it would be a good idea to do something will build flags so we can control whether this (and maybe other 'debug' code) is included in builds. We don't need this on the actual flight, so it would be good to be able to add and remove this functionality without having to make large changes to the codebase.

Esouder, 2022-10-05:
I can confirm that RTT is working on the flight computer. There were some issues where it didn't seem to be working, but that's just because the IMU task was fucked up and killing everything somehow.

[image]

With the IMU task removed, RTT works (at least for sending messages, which is all I need it for at the moment, so should be able to package this up cleanly pretty quick and merge it in.

I also made a very minor update the RTT documentation about where it's located on windows.

Esouder, 2022-10-05:
Moved it into the HAL to make things nicer for x86. Will test again, should be good to go after that. 

Am I jinxing myself? of course. Do I care? Of course not. Will I suffer for my hubris? Absolutely.

Esouder, 2022-10-05:
Tested it, still works fine.

PR #67 (2022-11-05): Rtt
## Description

Copied @davinbirdi's work from the DAQ board, got RTT working (board->pc, at least) on the flight computer. 

Fixes #50 

# Checklist:

- [x] My code follows the [style guidelines](https://github.com/UBC-Rocket/Whistler-Blackcomb-v2/blob/master/doc/WritingCleanCode.md) of this project
- [x] I have performed a self-review of my code
- [x] I have commented my code, particularly in hard-to-understand areas
- [x] I have made any necessary changes to the documentation
- [x] I have added tests, where possible, that prove this change is effective or that this feature works
   - n/ [... cut]
closed 2022-11-05 by Esouder